DronesForBlood/DronesForBlood #15: DroneLink - Recieve and send communication to the drone

Crowdedlight, 2018-09-03:
- [ ] Receive pos
- [ ] update mission waypoints
- [ ] Get next waypoint and endpoint
- [ ] Sending commands like stop, stay pos.... and so on

Crowdedlight, 2018-09-13:
On Hold, waiting on Mavlink_lora module.

Crowdedlight, 2018-09-29:
Extended mavlink lora module. Can now upload missions. Waiting on feedback from Kjeld before extending further with interface. Added list of cmds to implement interface for.

Crowdedlight, 2018-10-04:
Heartbeat implemented with 1hz output. Set timeout on pixhawk accordingly.

Crowdedlight, 2018-10-05:
Redo heartbeat, so lora just relays GCS heartbeat. So drone knows that if GCS_master fails, or isn't active, the heartbeat stops even if Lora is running

Crowdedlight, 2018-10-29:
Implement Following interfaces with high priority

- [ ] Mission download
- [ ] MISSION_CURRENT
- [ ] MISSION_ITEM_REACHED <-- Does this output on px4 for each waypoint, or just for mission end?

Crowdedlight, 2018-11-15:
MISSION_ITEM_REACHED <-- Only outputs at end of mission

Crowdedlight, 2018-11-15:
DO_CHANGE_HEIGHT

Crowdedlight, 2018-11-22:
Do change height and nav follow and change position doesn't seem to be implemented in px4 stack.

jlrandulfe, 2018-11-29:
Is this issue over? If there are new requirements/bugs regarding the dronelink, maybe they can be created in new issues

Crowdedlight, 2018-11-29:
Only think not done so-far we might want is mission download. However not sure if we plan on doing that. Otherwise we can create a new issue with that.

closed 2018-11-29 by Crowdedlight