DronesForBlood/DronesForBlood #29: The Gazebo simulation does not propagate some error messages through the mavlink

jlrandulfe, 2018-11-14:
There are 2 error messages at least that are not sent through the ros topic, when an acknowledge is required. The messages are the following:

ERROR [mavlink] ERROR: wp index out of bounds

Crowdedlight, 2018-11-14:
Do the simulator console say that data-link is lost ? 
If the heartbeats isn't sent you won't get the status messages

jlrandulfe, 2018-11-14:
The heartbeat is ok, and I am receiving other ack messages. But those errors are prompted in the Gazebo terminal and they are not propagated to the GCS master through mavlink. At least not in the response to the ack command (It answers ack is ok)

Crowdedlight, 2018-11-14:
Those messages do not get output in any ack messages or as part of the ack messages from PX4. Those get sent though the statuslink topic. As this is PX4 logging with extra information. So you will get them by listen to "mavlnk_statustext" topic. This is how PX4 implemented it. We can't change that

closed 2018-11-14 by Crowdedlight