DronesForBlood/DronesForBlood #30: mavlink does not upload a second mission if one is already sent

jlrandulfe, 2018-11-15:
When sending a second mission, after reaching a waypoint, the mavlink does not upload the new mission to the drone. The log message obtained is:

    [ INFO] [1542299256.068863962, 380.594000000]: Already sent this sequence number once

Should an extra command be sent before uploading a second mission?

Crowdedlight, 2018-11-15:
This sounds like it could happen if trying to upload new mission, while it is already uploading a mission. 

It is normal to get some ``Already sent this sequence number once`` on a normal mission upload, as the timeout on the px4 implementation isn't good for long range telemetry. So you will probably see some of these messages in normal operations. 

But it should still upload the mission, especially after timeout hits if it doesn't get asked for next item. And give you an ``mission_result_approved``. 

So in what circumstances does it happen? Does it happen while uploading another mission already?

jlrandulfe, 2018-11-15:
No, it happens after the first mission is uploaded. Any newer tries raises this message, and no mission is uploaded. Until the end of execution. 

The second try is sent around 8 seconds later, and then everytime after 3 seconds. Always the same answer, and no new mission is uploaded

Crowdedlight, 2018-11-16:
Hmm. That is quite odd. I have used the python script to upload missions
multiple times in the same mavlink instance before. I'll look into it
tomorrow. Must be a issue with resetting the counter.

Crowdedlight, 2018-11-16:
I don't see any issues in the code, and I know one of the other groups uses this code with no issues. Same code we have used to test missions earlier with uploading multiple missions while flying to check behaviour. 

I think the error lies somewhere else. Can you post full log from the mavlink_lora commandline when the error occurs? 

A possible error could be if you send mission items with wrong ``seq``, as error messages occurs when same ``seq`` is requested multiple times. I assume you remember to set sequence to start from 0, and increment per mission_item, everytime you upload a mission.

Otherwise we can check it out on monday.

jlrandulfe, 2018-11-22:
The sent mission was starting with a sequence number of 1, and hence it was not accepted by the drone. After correcting that, the code works

closed 2018-11-22 by jlrandulfe