DronesForBlood/DronesForBlood #32: Publish a heartbeat from the UserLink and Pathplanner nodes

jlrandulfe, 2018-11-26:


Crowdedlight, 2018-11-28:
Heartbeat is given in format:
```c++
uint8 id //Id of the node. Chose one to give the node. The monitor uses id + name to recognize nodes
string name //name of node
float32 expected_interval //expected interval to receive heartbeats from the node. given in seconds float
string main_status //main status, like "Running, Error, Rebooting, Crashed and so on"
string current_task //current task the node is performing. To monitor what the nodes is doing
bool has_error //flag to set if node has detected error itself, but is still sending heartbeats. 
```
If Lora_ground_control is present on your system the message format can be from it. The msg is called: "heartbeat_node.msg" under the lora_ground_control package.

Crowdedlight, 2018-12-05:
Topic: `/lora_ground_control/heartbeat_nodes_rx`

Crowdedlight, 2018-12-05:
Implemented heartbeats on UTM node 035425cedccc40cb57b7ae56a2b4fe3b67e87a67

jlrandulfe, 2018-12-08:
Added heartbeat from GCS master. Tested with lora monitor node

jlrandulfe, 2018-12-08:
Added hearbeat from docking station. However, the TIMEOUT of lora should be lowered down, as it demands slow nodes to work much faster, and hence increase the GCS workload

jlrandulfe, 2018-12-08:
I made a typo. The timeout should be raised

closed 2018-12-09 by jlrandulfe