DronesForBlood/DronesForBlood #34: Head towards a rally point if the battery is very low

jlrandulfe, 2018-11-26:
The GCS master has to communicate to the pathplanner if there is an emergency situation. If that happens, the pathplanner has to create a path that ends in the rally point.

jlrandulfe, 2018-11-26:
When an emergency situation, a decision has to be made:

- Go to the closest rally point

- Go to the docking station, as the drone is still able to get there

- land where it is now

jlrandulfe, 2018-12-06:
@Crowdedlight shall we dismiss this issue? It'd be a good feature, but I think the drone firmware gives enough protection in case on an unexpected drop down of the battery level

Crowdedlight, 2018-12-06:
For the theoretic/ideal system it should be handled, as the utm gives rally points. So you can navigate to a safe area while having battery to move around. 

But for our timeframe and monday, lets drop it. Just have to describe it in the report and how we would have done it. So it shows we have thought about it, but didn't manage to get it done

F1nsk, 2018-12-06:
I have done a bare bones i intergration, where  doesnt peiorities where it should be heading

jlrandulfe, 2018-12-06:
@F1nsk that was more focussed for #53, but it's true that we can addapt the code for that for this issue as well. I will think about it when I have some time tomorrow

closed 2018-12-08 by jlrandulfe