DronesForBlood/DronesForBlood #38: Error is raised sometimes when executing UTM

jlrandulfe, 2018-11-27:
```python
Traceback (most recent call last):
  File "/home/jalop17/01-scripts/DronesForBlood/software/gcs/devel/lib/python2.7/dist-packages/utm/msg/_utm_tracking_data.py", line 109, in serialize
    buff.write(_get_struct_2q2d4q2d5q().pack(_x.uav_id, _x.uav_op_status, _x.pos_cur_lat_dd, _x.pos_cur_lng_dd, _x.pos_cur_alt_m, _x.pos_cur_hdg_deg, _x.pos_cur_vel_mps, _x.pos_cur_gps_timestamp, _x.wp_next_lat_dd, _x.wp_next_lng_dd, _x.wp_next_alt_m, _x.wp_next_hdg_deg, _x.wp_next_vel_mps, _x.wp_next_eta_epoch, _x.uav_bat_soc))
struct.error: required argument is not an integer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 1066, in publish
    serialize_message(b, self.seq, message)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/msg.py", line 152, in serialize_message
    msg.serialize(b)
  File "/home/jalop17/01-scripts/DronesForBlood/software/gcs/devel/lib/python2.7/dist-packages/utm/msg/_utm_tracking_data.py", line 110, in serialize
    except struct.error as se: self._check_types(struct.error("%s: '%s' when writing '%s'" % (type(se), str(se), str(locals().get('_x', self)))))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/message.py", line 334, in _check_types
    check_type(n, t, getattr(self, n))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/message.py", line 206, in check_type
    raise SerializationError('field %s must be an integer type'%field_name)
genpy.message.SerializationError: field pos_cur_alt_m must be an integer type

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jalop17/01-scripts/DronesForBlood/software/gcs/src/utm/src/utm_node.py", line 380, in <module>
    utm_node.run()
  File "/home/jalop17/01-scripts/DronesForBlood/software/gcs/src/utm/src/utm_node.py", line 98, in run
    self.fetch_tracking_data()
  File "/home/jalop17/01-scripts/DronesForBlood/software/gcs/src/utm/src/utm_node.py", line 251, in fetch_tracking_data
    self.tracking_data_pub.publish(msg)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 886, in publish
    raise ROSSerializationException(str(e))
rospy.exceptions.ROSSerializationException: field pos_cur_alt_m must be an integer type
```

andkgl, 2018-12-03:
Should be fixed. It never crashes for me.

jlrandulfe, 2018-12-03:
Reopening the issue until the reason of the crash at #66 is clarified

andkgl, 2018-12-04:
As described in the other issue, this is most likely an error in GCS.
The GCS doesn't set the message up correctly and it therefore never reaches UTM.

jlrandulfe, 2018-12-05:
Fixed. The formad sent from GCS master was incorrect

closed 2018-12-05 by jlrandulfe