DronesForBlood/DronesForBlood #40: going into state fly multiple times while flying. Is this planned behaviour?

Crowdedlight, 2018-11-28:
going into FSM state of fly multiple times while flying mission with no data link lost. Is this planned behaviour?

```
[INFO] [1543399948.986284]: FSM state: take_off
[INFO] [1543399961.836375]: FSM state: fly
[INFO] [1543399961.846478]: FSM state: upload_mission
[ INFO] [1543399961.853806113]: New mission. Length:1
[ INFO] [1543399961.853904414]: Sending mission count
[ INFO] [1543399962.424018933]: Next item asked for:0
[ INFO] [1543399962.493943107]: Already sent this sequence number once
[ INFO] [1543399962.593972211]: MAV_MISSION_ACCEPTED
[INFO] [1543399962.594480]: Mission upload accepted
[INFO] [1543399962.596331]: FSM state: start_mission
[ INFO] [1543399962.603781888]: Sending Command_long with id: 176, and confirmation: 0
[ INFO] [1543399962.603926370]: Sending Command_long with id: 300, and confirmation: 0
[ INFO] [1543399964.113855199]: Sending Command_long with id: 300, and confirmation: 1
[ INFO] [1543399965.623781255]: Sending Command_long with id: 300, and confirmation: 2
[ INFO] [1543399965.863857539]: MAV_RESULT_ACCEPTED
[INFO] [1543399965.866334]: FSM state: fly
[INFO] [1543399965.876631]: FSM state: upload_mission
[ INFO] [1543399965.883848502]: New mission. Length:1
[ INFO] [1543399965.883932659]: Sending mission count
[ INFO] [1543399966.153987482]: Next item asked for:0
[ INFO] [1543399966.293899972]: MAV_MISSION_ACCEPTED
[INFO] [1543399966.294411]: Mission upload accepted
[INFO] [1543399966.296349]: FSM state: start_mission
[ INFO] [1543399966.303781648]: Sending Command_long with id: 300, and confirmation: 0
[ INFO] [1543399966.303967764]: Sending Command_long with id: 176, and confirmation: 0
[ INFO] [1543399967.813875602]: Sending Command_long with id: 176, and confirmation: 1
[ INFO] [1543399969.323780375]: Sending Command_long with id: 176, and confirmation: 2
[ INFO] [1543399969.573881456]: MAV_RESULT_ACCEPTED
[ INFO] [1543399971.313809715]: Sending Command_long with id: 300, and confirmation: 0
[ INFO] [1543399971.314033386]: Sending Command_long with id: 176, and confirmation: 0
[ INFO] [1543399971.643821522]: MAV_RESULT_ACCEPTED
[ INFO] [1543399971.643915350]: MAV_RESULT_ACCEPTED
[INFO] [1543399971.646363]: FSM state: fly
[INFO] [1543399980.306360]: FSM state: land
[ INFO] [1543399985.323834350]: Sending Command_long with id: 21, and confirmation: 0
[ INFO] [1543399985.543893932]: MAV_RESULT_ACCEPTED
[INFO] [1543400039.994514]: Communication recovered
[INFO] [1543400057.846430]: FSM state: start
[ INFO] [1543400057.853833912]: Sending Command_long with id: 400, and confirmation: 0
[ INFO] [1543400057.974000488]: MAV_RESULT_ACCEPTED
[INFO] [1543400057.974493]: Drone is disarmed
```

jlrandulfe, 2018-11-28:
Yes, it is how the FSM is designed. Every time a new waypoint is reached, the evolution of the states is FLY --> UPLOAD_MISSION --> START_MISSION --> FLY

There is a states diagram and more info in [the wiki](https://github.com/DronesForBlood/DronesForBlood/wiki/GCS%3A-Master-node-Finite-State-Machine)

Crowdedlight, 2018-11-28:
But it still continues flying to next waypoint when in upload mission?

jlrandulfe, 2018-11-28:
Yes! The mission is uploaded again when the pathplanner finish the calculations, and then the start mission command is sent. In any case, the mission execution is never paused or canceled

closed 2018-11-28 by jlrandulfe