DronesForBlood/DronesForBlood #46: Get drone groundspeed and publish it, so it can be registered in the UTM

jlrandulfe, 2018-11-29:


Crowdedlight, 2018-11-29:
publish `GPS_RAW_INT` message, as it also give GPS fix that can be used in ground_control for stability in drawing drone path. Also gets "Course-over-ground", that isn't heading, but the current course over the ground. not sure if we can use that or even need it. 

https://mavlink.io/en/messages/common.html#GPS_RAW_INT

Crowdedlight, 2018-11-29:
@jlrandulfe output will be on topic `mavlink_gps_raw`

Format will be this msg: 
```
uint64 time_usec
uint8 fix_type
int32 lat
int32 lon
int32 alt
uint16 eph
uint16 epv
uint16 vel
uint16 cog
uint8 satellites_visible
```

Same format as in the common link. Groundspeed is given in cm/s in the "vel" parameter. 

If what he want is course over ground, which is given in `cdeg` you can get that directly from the `cog` parameter. (It is not heading, but direction of movement)

Crowdedlight, 2018-11-29:
Do note that altitude is in **mm** if you ever plan on using it. And it is MSL altitude.

Crowdedlight, 2018-11-29:
I changed some of the unit. This shows what it gives correctly:
```cpp
uint64 time_usec //us
uint8 fix_type
int32 lat //dd
int32 lon //dd
int32 alt //m
uint16 eph
uint16 epv
uint16 vel //m/s
uint16 cog //deg
uint8 satellites_visible
```

Crowdedlight, 2018-11-29:
Sent latest mavlink_lora with groundspeed. This issue should be solved @jlrandulfe

closed 2018-12-03 by Crowdedlight