DronesForBlood/DronesForBlood #50: UX communication

F1nsk, 2018-12-01:
`[WARN] [1543667719.197863, 346.390000]: Could not process inbound connection: topic types do not match: [std_msgs/Bool] vs. [std_msgs/Int16MultiArray]{'message_definition': 'bool data', 'callerid': '/gcs_master', 'tcp_nodelay': '0', 'md5sum': '8b94c1b53db61fb6aed406028ad6332a', 'topic': '/dronelink/start', 'type': 'std_msgs/Bool'}
`

jlrandulfe, 2018-12-03:
That is a typo. The name should be `userlink/start`. It was not changing for not altering the compatibility with the UserLink node, but it should definitely be changed.

Gilsaa, 2018-12-03:
Topic prefix is changed to "userlink/" in newest commit of userlink.

jlrandulfe, 2018-12-04:
Topic names in GCS master are changed now in the userlink branch. @Gilsaa, please test it with the userlink node, as I still have some troubles for running it in my laptop (I think I know why it doesn't run in my PC, I will fix it soon)

F1nsk, 2018-12-05:
Seems to be working now,  the error is the following, but this is due to a missing implementation? 
`[ERROR] [1543996069.792967, 48.360000]: [Client 0] [id: advertise:/dronelink/destination:2] advertise: Unable to load the manifest for package dronelink. Caused by: dronelink
`

closed 2018-12-05 by jlrandulfe