DronesForBlood/DronesForBlood #66: Crash and fall from 50m height.

Crowdedlight, 2018-12-03:
Console output from the exception/crash. We still do not know why it went out of mission mode and cut all motors. 

```
[ INFO] [1543841049.223782400]: Opening serial device: /dev/ttyUSB0 baudrate 57600
[ INFO] [1543841049.224584414]: Serial Port initialized
[INFO] [1543841050.096847]: FSM state: start
[WARN] [1543841050.098378]: Low battery level: 0%. Take off not allowed
[WARN] [1543841050.099534]: WARNING: Very low battery level: 0%
[INFO] [1543841050.405364]: Communication recovered
Connecting to UTM node
Succesfully connected to UTM node
Setting up subscribers and publishers
Subscribes and publishers succesfully setup
[INFO] [1543841178.897151]: FSM state: planner_setup
[INFO] [1543841178.899870]: ACTIVATE PLANNER is True
The pathplanner is ready
Generating a new map
Start: 55.4721 10.4147    Goal: 55.4722 10.4168    Node dist: 1    Map width: 200    Pad length: 100    Map size: 201 x 331
[INFO] [1543841179.881502]: FSM state: arm
[ INFO] [1543841179.932780154]: Sending Command_long with id: 400, and confirmation: 0
[ INFO] [1543841179.932956711]: Sending Command_long with id: 22, and confirmation: 0
[ INFO] [1543841181.111834367]: MAV_RESULT_ACCEPTED
[ INFO] [1543841181.211830174]: MAV_RESULT_ACCEPTED
[INFO] [1543841181.215035]: FSM state: take_off
[INFO] [1543841181.269737]: Current path LENGHT: 1
Traceback (most recent call last):
  File "/home/crow/workspace/catkin_ws/devel/lib/python2.7/dist-packages/utm/msg/_utm_tracking_data.py", line 109, in serialize
    buff.write(_get_struct_2q5dq5d2q().pack(_x.uav_id, _x.uav_op_status, _x.pos_cur_lat_dd, _x.pos_cur_lng_dd, _x.pos_cur_alt_m, _x.pos_cur_hdg_deg, _x.pos_cur_vel_mps, _x.pos_cur_gps_timestamp, _x.wp_next_lat_dd, _x.wp_next_lng_dd, _x.wp_next_alt_m, _x.wp_next_hdg_deg, _x.wp_next_vel_mps, _x.wp_next_eta_epoch, _x.uav_bat_soc))
struct.error: required argument is not an integer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 882, in publish
    self.impl.publish(data)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 1066, in publish
    serialize_message(b, self.seq, message)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/msg.py", line 152, in serialize_message
    msg.serialize(b)
  File "/home/crow/workspace/catkin_ws/devel/lib/python2.7/dist-packages/utm/msg/_utm_tracking_data.py", line 110, in serialize
    except struct.error as se: self._check_types(struct.error("%s: '%s' when writing '%s'" % (type(se), str(se), str(locals().get('_x', self)))))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/message.py", line 334, in _check_types
    check_type(n, t, getattr(self, n))
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genpy/message.py", line 206, in check_type
    raise SerializationError('field %s must be an integer type'%field_name)
genpy.message.SerializationError: field pos_cur_gps_timestamp must be an integer type

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/crow/workspace/catkin_ws/src/gcs_master/nodes/gcs_master", line 29, in <module>
    gcs_master.main(args.altitude, args.batt1, args.batt2, args.hovertime)
  File "/home/crow/workspace/catkin_ws/src/gcs_master/src/gcs_master/gcs_master.py", line 443, in main
    gcs_master.run()
  File "/home/crow/workspace/catkin_ws/src/gcs_master/src/gcs_master/gcs_master.py", line 425, in run
    self.send_utm_data()
  File "/home/crow/workspace/catkin_ws/src/gcs_master/src/gcs_master/gcs_master.py", line 399, in send_utm_data
    self.utm_data_pub.publish(msg)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 886, in publish
    raise ROSSerializationException(str(e))
rospy.exceptions.ROSSerializationException: field pos_cur_gps_timestamp must be an integer type
[gcs_master-2] process has died [pid 348, exit code 1, cmd /home/crow/workspace/catkin_ws/src/gcs_master/nodes/gcs_master --batt1 40 --batt2 10 --altitude 50 --hovertime 20 __name:=gcs_master __log:=/home/crow/.ros/log/210a949e-f6f7-11e8-96b4-302432c9df97/gcs_master-2.log].
log file: /home/crow/.ros/log/210a949e-f6f7-11e8-96b4-302432c9df97/gcs_master-2*.log
^C[utm-4] killing on exit
[pathplanner-3] killing on exit
[mavlink_lora-1] killing on exit
Closing visualizer
Closing visualizer complete
shutting down processing monitor...
... shutting down processing monitor complete
done
```

F1nsk, 2018-12-03:
How much damage did the drone sustain? 
Is irlock and sf10/a still working?

jlrandulfe, 2018-12-03:
The UTM system was the one crashing at first, and it propagated the crash afterwards to all of the systems.

That's the error that we have been observing in the simulation these days before, reported in #38 ... did you try simulating with the code you used before the test?

I can not figure out why the motors were cancelled out

jlrandulfe, 2018-12-03:
I observe that the bug in #83 may have already been fixed, but the updates are not in the develop branch (they are only in the path-planner branch), and the code in develop was still buggy.

Crowdedlight, 2018-12-03:
I think if your master sends disarm due to restart while the drone is in
the air. It works as a killswitch. And disarms.

I suggest making the master never send disarm, but use mission land way
point. As when mission is done with a land way point it auto disarms.

Also other groups have said they could only get precision land to work when
in mission.

This change would save us issues later if restarting master while flying.
While ensure we do precision land.

jlrandulfe, 2018-12-03:
> I think if your master sends disarm due to restart while the drone is in the air. It works as a killswitch. And disarms. I suggest making the master never send disarm, but use mission land way [...]

Can you check that in the log? I don't see how the master can be restarted, as there is no way in the code for this to happen,  and once and exception is raised the program exits

Crowdedlight, 2018-12-03:
Cannot see it in the log. Just thinking in regards to acceptance test. If
exceptions happen and we need a restart. I don't know if that happens. But
it is an easy fix to use a mission land and avoid the disarm possibility.

Also as we can't verify the land command works with precision landing. But
we know it works with mission.

jlrandulfe, 2018-12-03:
If the gcs_master is launched again after it is exited it can actually disarm the drone, that's true. I will work on that fix.

jlrandulfe, 2018-12-04:
But then the UTM node did crash because of an incorrect inccoming message format. And if that's the reason, it's good news, as it has an easy solution

andkgl, 2018-12-04:
The sending node crashes before the message is send, so that that it is never received. The UTM therefore never got it. (This is based purely on my experiences before and I might be wrong). But yea, it's an easy fix!

jlrandulfe, 2018-12-04:
New method for landing has been coded. Will be included in the next merge to develop branch

closed 2018-12-04 by jlrandulfe