DronesForBlood/DronesForBlood #76: mavlink issue?

F1nsk, 2018-12-05:
```
[ INFO] [1544017496.054297187, 165.100000000]: Sending Command_long with id: 176, and confirmation: 0
[ INFO] [1544017496.054416951, 165.100000000]: Sending Command_long with id: 300, and confirmation: 0
[ INFO] [1544017496.066923873, 165.110000000]: MAV_RESULT_ACCEPTED
[ INFO] [1544017496.103822293, 165.120000000]: MAV_RESULT_DENIED
[WARN] [1544017496.104410, 165.120000]: Command 300 denied
[WARN] [1544017496.113060, 165.122000]: MAV_RESULT_DENIED

```

Crowdedlight, 2018-12-05:
It accepts arming, and then denies mission_start (id 300). I assume there might not have been a valid mission. But how to handle it is up to @jlrandulfe

jlrandulfe, 2018-12-05:
If a command is denied, the code tries again. If you don't see more messages saying "Command denied" means that it was sent again successfully.

closed 2018-12-05 by jlrandulfe