DronesForBlood/DronesForBlood #77: GCS master issue

F1nsk, 2018-12-05:
```
  self.activate_planner_pub.publish(msg)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/topics.py", line 886, in publish
    raise ROSSerializationException(str(e))
rospy.exceptions.ROSSerializationException: <class 'struct.error'>: 'required argument is not a float' when writing 'header: 
  seq: 1
  stamp: 
    secs: 0
    nsecs:         0
  frame_id: ''
time_usec: 0
lat: None
lon: None
alt: 0.0
relative_alt: 0.0
heading: 0.0'
[gcs_master-2] process has died [pid 13475, exit code 1, cmd /home/finsk/catkin_ws/DronesForBlood/software/gcs/src/gcs_master/nodes/gcs_master --batt1 80 --batt2 20 --altitude 50 --hovertime 10 __name:=gcs_master __log:=/home/finsk/.ros/log/a21a05e6-f89c-11e8-9cf1-9c4e3636e5a8/gcs_master-2.log].
log file: /home/finsk/.ros/log/a21a05e6-f89c-11e8-9cf1-9c4e3636e5a8/gcs_master-2*.log

```

jlrandulfe, 2018-12-06:
I need more information about the error

jlrandulfe, 2018-12-06:
Did not see the error again. Closing the issue

closed 2018-12-06 by jlrandulfe